matrix-io/matrix-creator-malos #8: Node.Js only everloop demo worked from the docu

MexXxo, 2016-09-24:
Hi,

today I tired to run all the NodeJs Demos.
I'm using Node 6.6.0, I have installed zmq and protobuffjs (btw. please add to the documentation that this must be installed, or even better at a package.json than you can simply do an npm install). 

Also I needed to clone seperatly the protocol-bufers repo, here I didn't used the pointer from the malos repo, instead I took the latest master. 

The everloop demo worked! :)
But the rest of the Demos:
uv, humidity, pressure and imu just returned 0 values.
Zigbee I haven't tested.

Node.Js Logs:

node test_uv.js
Sending pings every 5 secondsMessage received: UV error: 0, Invalid configuration for UV driver.
Message received: UV error: 0, Invalid configuration for UV driver.
{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }

node test_humidity.js
Sending pings every 5 secondsMessage received: Humidity error: 0, Invalid configuration for Humidity driver.
Message received: Humidity error: 0, Invalid configuration for Humidity driver.
{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }

node test_everloop.js

node test_pressure.js
Sending pings every 5 seconds{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }

node test_imu.js
Sending pings every 5 seconds{ yaw: 0, pitch: 0, roll: 0 }
{ yaw: 0, pitch: 0, roll: 0 }
{ yaw: 0, pitch: 0, roll: 0 }
{ yaw: 0, pitch: 0, roll: 0 }
{ yaw: 0, pitch: 0, roll: 0 }

Logs from MALOS:

**************
MALOS starting
**************

You can query specific driver info using port 20012.
Registered driver IMU with port 20013.
Registered driver Humidity with port 20017.
Registered driver Everloop with port 20021.
Registered driver Pressure with port 20025.
Registered driver UV with port 20029.
Invalid configuration for UV driver.
Invalid configuration for UV driver.
New delay between updates for UV is 1000 ms.
New timeout after last ping for UV 6000 ms.
Invalid configuration for Humidity driver.
Invalid configuration for Humidity driver.
New delay between updates for Humidity is 1000 ms.
New timeout after last ping for Humidity 6000 ms.
New delay between updates for Pressure is 1000 ms.
New timeout after last ping for Pressure 6000 ms.
New delay between updates for IMU is 1000 ms.
New timeout after last ping for IMU 6000 ms.
New delay between updates for IMU is 2000 ms.
New timeout after last ping for IMU 6000 ms.

eighteyes, 2016-09-24:
It's very likely the format of the protobufs changed and invalidated the demos. Let us investigate.

nebiljabari, 2016-09-26:
Same here except for `test_imu.js` - Snippet:
```
Sending pings every 5 seconds{ yaw: 2.1193561742413096e-17,
  pitch: -47.79149627685547,
  roll: -2.8932242340087535e+25 }
{ yaw: 2.116101714839281e-17,
  pitch: -0.0007320902077481151,
  roll: -1.8954033148346924e+30 }
{ yaw: 5.288208669443359e-18,
  pitch: -7.135988653317327e-7,
  roll: -2.8961824000052737e+25 }
```

arhuaco, 2016-09-27:
"Message received: UV error: 0, Invalid configuration for UV driver. Message received: UV error: 0, Invalid configuration for UV driver." You shouldn't see this. I'll take a look early tomorrow to see what can be causing this issue.

arhuaco, 2016-09-27:
@MexXxo About ZigBee, we are testing with Bulbs but this is under-documented and you will need to run a program (and a new version of MALOS) to make this work. Do you have a ZigBee bulb?

MexXxo, 2016-09-27:
@arhuaco no I haven't tested ZigBee, currently I have also no device to test ZigBee with.

arhuaco, 2016-09-27:
@MexXxo 

Thanks for the suggestions.

I just added the package.json file:

https://github.com/matrix-io/matrix-creator-malos/commit/6810d4b8a9d0741f3f0a77fdae566d2ce6bae42e

I also removed the code that generates the following messages. We were sending invalid configurations to test the code that process them but this code is already working well.

"Invalid configuration for UV driver. Message received: UV error: 0, Invalid configuration for UV driver"

For humidity I get:

{ humidity: 74.72823333740234, temperature: 35.519264221191406 }
{ humidity: 74.7742691040039, temperature: 35.51971435546875 }
{ humidity: 74.83812713623047, temperature: 35.5203857421875 }

Can you please tell us what version of MALOS you're running? You can try the next command to upgrade to the latest version:

apt-get update && matrix-creator-malos

Let us know...

nebiljabari, 2016-09-28:
@arhuaco how to know the MALOS version ? The malos command line don't take argument right ? Ive tried : `malos help`, `malos -v`, `malos version` no success

I have re-cloned the full `matrix-creator-malos` folder
done `git submodule init && git submodule update` and `node install` for the package.json
...same result :-/ 

```
Sending pings every 5 secondsMessage received: Humidity error: 0, Invalid configuration for Humidity driver.
Message received: Humidity error: 0, Invalid configuration for Humidity driver.
{ humidity: 0, temperature: 0 }
```
Nelson, maybe I missed a step but when I try to `sudo apt-get matrix-creator-malos` I get an error (translate from French) : `E: matrix-creator-malos operator is not valid`

arhuaco, 2016-09-28:
@nebiljabari 

Have you followed the getting started page?

https://github.com/matrix-io/matrix-creator-quickstart/wiki/2.-Getting-Started

The second line of the following command should tell you the version that you have installed.

apt-cache show matrix-creator-malos

ATM the MALOS binary doesn't print the version but it should. Something to do later. ATM we're managing versions with Debian packages only.

nebiljabari, 2016-09-28:
Yes Nelson already done the Getting Started page. 
When I try to re-install : `sudo apt-get install matrix-creator-init matrix-creator-malos`
The output tell me that I already have the most recent version

After `apt-cache show matrix-creator-malos` I get : `Version: 0.1-2`
You want the full output ?

arhuaco, 2016-09-28:
I have the same version. No full output needed. You should no longer see the "Invalid configuration for Humidity driver" as I removed this test and the commit is already in the master branch.

https://github.com/matrix-io/matrix-creator-malos/commit/b5f3874a391b7c64bf7f7eaafbd19735450274f2

We can try programming the MCU again. Please try this:

    sudo rm /usr/share/admobilize/matrix-creator/sam3-program.bash.done
    sudo /usr/share/admobilize/matrix-creator/sam3-program.bash

and then restart MALOS and run the following command again.

    node test_humidity.js

arhuaco, 2016-09-28:
If that doesn't work, you can try the same as above but:

     sudo /usr/share/admobilize/matrix-creator/creator-init.bash 

instead of:

    sudo /usr/share/admobilize/matrix-creator/sam3-program.bash

nebiljabari, 2016-09-28:
Thanks @arhuaco your code just rock my Matrix. I ran all the test (except zigbee) and they work

I followed your first lines :

> We can try programming the MCU again. Please try this:
> ```
> sudo rm /usr/share/admobilize/matrix-creator/sam3-program.bash.done
> sudo /usr/share/admobilize/matrix-creator/sam3-program.bash
> ```

arhuaco, 2016-09-28:
Good to hear @nebiljabari .

This is an issue we have. Opening https://github.com/matrix-io/matrix-creator-init/issues/4 .

Have fun!

MexXxo, 2016-10-19:
Hi @arhuaco, 
I would like to reopen this issue. I have upgraded and updated everything, but only the everloop is working from the node tests ...
All other show no real values, all of them are using protocol buffers

nebiljabari, 2016-10-19:
@MexXxo what do mean by :

> All other show no real values

MexXxo, 2016-10-22:
Hi @nebiljabari 
so all my sensors are returning zero values, see below.
I have the latest malos, latest protocolbuffers, tried to reprogram the MCU.
So I have no idea, why all my sensors are returning no data.
However the mic and everloop e.g. are working fine!

 node test_humidity.js
Sending pings every 5 seconds{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }
{ humidity: 0, temperature: 0 }

node test_pressure.js
Sending pings every 5 seconds{ pressure: 0, altitude: 0, temperature: 0 }
{ pressure: 0, altitude: 0, temperature: 0 }

 node test_uv.js
Sending pings every 5 seconds{ uv_index: 0, oms_risk: 'Low' }
{ uv_index: 0, oms_risk: 'Low' }

  node test_imu.js
Sending pings every 5 seconds{ yaw: 180, pitch: 0, roll: 0 }

arhuaco, 2016-10-22:
Please try:
    sudo rm /usr/share/admobilize/matrix-creator/sam3-program.bash.done
    sudo reboot
to see if the problem persists.

MexXxo, 2016-10-22:
I have deleted the flag and restarted my RPi.
Result is the same (tested humidity, pressure and imu)

I'm running node 6.9.0

MexXxo, 2016-10-22:
I see that the protocol-buffers are on the commit 353527c. Should I use the latest master of the protocol buffers?

arhuaco, 2016-10-22:
The messages haven't changed but try using the latest commit from master. There is an open issue that might be affecting you: https://github.com/matrix-io/matrix-creator-init/issues/10 , we do need to improve initialization. You can try programming the MCU a few times without restarting to see if this fixes the issue.

    sudo rm -f /usr/share/admobilize/matrix-creator/sam3-program.bash.done
    sudo /usr/share/admobilize/matrix-creator/sam3-program.bash 
    sudo killall malos
    malos
    # test a driver

MexXxo, 2016-10-22:
No success :( 
Maybe I will first try to confirm that the sensors are running on HAL Layer ?
Maybe it could also be helpful, to do a clean deinstall and reinstall of the matrix-creator components

arhuaco, 2016-10-22:
You can try installing everything again but I'm not sure it will work. I'll ask @eljuguetero to see if he has an idea of what to do.

HAL is linked into MALOS so by trying MALOS you're also trying HAL. So far MALOS always works when HAL is working. Please also check that the board is properly connected to the Pi (push it...).

Can you try upgrading MALOS? This is the closest we can be to a reinstall.

    sudo apt-get update
    sudo apt-get upgrade
    apt-get install matrix-creator-malos matrix-creator-init
    sudo reboot

MexXxo, 2016-10-22:
Thanks @arhuaco,
still no success, I needed to use sudo apt-get install matrix-creator-malos matrix-creator-init
But I had already the newest versions ... 

Also like you said, the HAL is also returning the same zero values like the node tests.

MexXxo, 2016-10-22:
Hey @arhuaco, 
it's working right now :)

I did the following things:
completely removed the matrix creator from my RPi and again put it on my RPi (of course I shutted it down before)
and I also did an rpi-update (but without restart, so it seems to be the remove and set from the matrix creator)

Thanks for your effort @arhuaco !

arhuaco, 2016-10-22:
Perhaps it was a bad contact. I'm glad you can use the Creator now! :+1: 

FYI we're adding more documentation to MALOS...

https://github.com/matrix-io/matrix-creator-malos/

MexXxo, 2016-10-22:
Update:
So the temperature from the sensors says something between 31-34. 
In Celsius and Fahrenheit this seems to be very unrealistic (it should be round about 22°C here)
Also the values of pressure humidity and imu are not changing their values... (others I have not tested)

node test_humidity.js
Sending pings every 5 seconds{ humidity: 73.66251373291016, temperature: 34.37471008300781 }

node test_pressure.js
Sending pings every 5 seconds{ pressure: 100498, altitude: 65.9375, temperature: 31.8125 }

MexXxo, 2016-10-22:
With a little bit testing it looks like the values are only grabbed on a system start, but not updating.

@arhuaco can you confirm this?

arhuaco, 2016-10-22:
I have never seen this one. I'll have to defer this issue to more people. AFAIK HAL grabs the values every time.

ATM components in the board are affecting the temperature you're reading so the values you're seeing make sense. I asked here and all the values make sense. We're working on a software fix for the temperature issue and it should be released soon.

MexXxo, 2016-10-22:
So imu, pressure and humidity are not updating.
The imu has not changed it's values also when I moved the board.
The imu values have changed after a reboot (I simply rebooted it holding it diagonally).

A MALOS only restart did not changed these values.

nebiljabari, 2016-10-29:
I updated today and all the .js test (except everloop) gave fxed values. I had to : 
`sudo /usr/share/admobilize/matrix-creator/sam3-program.bash` and we're back in business

@arhuaco I think you should put a note in the ReadMe file for others

MexXxo, 2016-11-12:
did an update && upgrade, rebooted pi and reprogrammed mcu. Finally the sensors are working :)

With the temperature (on both sensors) showing wrong values, I could also move the creator more away from the raspi right?

closed 2017-01-31 by brianofrokk3r