matrix-io/matrix-creator-malos #15: Address already in use at rPi lauch

nebiljabari, 2016-10-05:
After a bad kill of `~/matrix-os $ node index.js`. I `sudo reboot` and also `sudo poweroff` my Pi but the error persist. To avoid it I have to `sudo pkill`here is my procedure (on my Pi launch ):
```
pi@raspberrypi:~ $ malos
**************
MALOS starting
**************

You can query specific driver info using port 20012.
terminate called after throwing an instance of 'zmq::error_t'
  what():  Address already in use
Abandon
pi@raspberrypi:~ $ pkill -9 malos
pkill: killing pid 425 failed: Opération non permise // = not authorized Operation (/or/ Process)
pi@raspberrypi:~ $ sudo pkill -9 malos
pi@raspberrypi:~ $ malos
**************
MALOS starting
**************

You can query specific driver info using port 20012.
Registered driver IMU with port 20013.
Registered driver Humidity with port 20017.
Registered driver Everloop with port 20021.
Registered driver Pressure with port 20025.
Registered driver UV with port 20029.
Registered driver ZigbeeBulb with port 20033.
Registered driver MicArray_Alsa with port 20037.
Registered driver Lirc with port 20041.
```

matrix-creator-malos v 0.1-4.5

arhuaco, 2016-10-05:
@nebiljabari : With the latest update MALOS starts as a service so you no longer need to start it by hand. If you want debug output you can kill it and start it by hand as you just did.

nebiljabari, 2016-10-05:
I think you set a record @arhuaco ! Fastest closed issue on GitHub ;-)

closed 2016-10-05 by nebiljabari