matrix-io/matrix-creator-malos #20: Registered driver Aborted (core dumped)

buildog, 2016-10-14:
```
root@raspberrypi3-82d23ee:/usr/src/app# malos

**************                                                                                                                             
MALOS starting                                                                                                                             
**************                                                                                                                             
                                                                                                                                           
You can query specific driver info using port 20012.                                                                                       
pure virtual method called                                                                                                                 
terminate called without an active exception                                                                                               
Registered driver Aborted (core dumped)   
```

arhuaco, 2016-10-14:
Oh, I hadn't seen that one... I'll check the most recent version of MALOS again.

arhuaco, 2016-10-14:
Do you have the most recent version? 0.1-4.5
In any case, we hadn't seen this issue before and I don't know what can cause it.
This part of the code should be deterministic.

The first driver to be registered is IMU.

@eljuguetero Do you have an idea of what can be happening here?

buildog, 2016-10-14:
latest version, that's a fresh install

if I send several time the malos command, some time I got this one: 

```
**************                                                                                                                             
MALOS starting                                                                                                                             
**************                                                                                                                             
                                                                                                                                           
You can query specific driver info using port 20012.                                                                                       
Registered driver pure virtual method called                                                                                               
IMU with port 20013.                                                                                                                       
terminate called without an active exception                                                                                               
Aborted (core dumped)                                                                                                                      
root@raspberrypi3-82d23ee:/usr/src/app#                                                                                                                
```

From the OS script, I should have a loading led animation after `node ./script` (`Matrix.device.drivers.led.loader()`) but until now it's not lighting :-)

buildog, 2016-10-14:
```
root@raspberrypi3-82d23ee:/usr/src/app# apt-cache show matrix-creator-malos                                                                
Package: matrix-creator-malos                                                                                                              
Version: 0.1-4.5                                                                                                                           
Architecture: armhf                                                                                                                        
Maintainer: Admobilize <info@admobilize.com>                                                                                               
Installed-Size: 1677                                                                                                                       
Depends: libc6 (>= 2.17), libgcc1 (>= 1:4.4.0), libstdc++6 (>= 4.9)                                                                        
Filename: ./matrix-creator-malos_0.1-4.5_armhf.deb                                                                                         
Size: 461554                                                                                                                               
MD5sum: 2b33c3783f526455fa0b72612275d013                                                                                                   
SHA1: d398ff62af723bf61ffc24804b146feaf2d57eb4                                                                                             
SHA256: 5a00a9078d6beae5e9361848423720777546a2d4ee48a52ee609856d3eaf0f6f                                                                   
Section: embedded                                                                                                                          
Priority: optional                                                                                                                         
Homepage: https://github.com/matrix-io/matrix-creator-malos                                                                                
Description: Allows queries to MATRIX Creator hardware via ZMQ.                                                                            
 Matrix Creator (https://creator.matrix.one) is a board that can be added                                                                  
 to a Raspberry Pi v2 or v3. This board has many sensors. This program                                                                     
 allows other programs to access the board hardware via ZMQ sockets.                                                                       
Description-md5: 81858a60ae50a580b33529d209b664b1 
```

arhuaco, 2016-10-14:
I see. I thought more.

1) With a core dump the messages are not reliable because we're not flushing stderr after the initialization of each driver. We should do that.
Not 100% sure, but thinking I think an issue like the one you're having could happen if MALOS is started after services that are spamming it are running.

I'll compile a version for you to try. One that flushes stderr after each initialization. OK?
Compiling now.

buildog, 2016-10-14:
yeah let's try, ty @arhuaco

arhuaco, 2016-10-14:
Here's a version of MALOS with debugging symbols and more error messages (and delays).

[malos.gz](https://github.com/matrix-io/matrix-creator-malos/files/528533/malos.gz)

Here's what I changed:

[debug.patch.gz](https://github.com/matrix-io/matrix-creator-malos/files/528534/debug.patch.gz)

If it crashes we can run it with gdb and get more insight.

buildog, 2016-10-14:
Ok, I installed wget, ddl the file and gunzip it, but I don't know what's next :-)

arhuaco, 2016-10-14:
Ah! Just run it.

    ./malos

And let us know if it crashes (it should because of what you’re seeing) and what is printed last.

In addition, this can help us stop the issue:

    sudo apt-get install gdb
    gdb ./malos
    # Type run when a prompt is shown
    # Send us what is printed last when the program crashes.

Thanks!

buildog, 2016-10-14:
Great

```
MALOS starting                                                                                                                             
**************                                                                                                                             
                                                                                                                                           
You can query specific driver info using port 20012.                                                                                       
Starting initialization of  IMU                                                                                                            
Creating threads of  IMU                                                                                                                   
Creating config thread  IMU                                                                                                                
Creating keepalive thread  pure virtual method called                                                                                      
IMU                                                                                                                                        
terminate called without an active exception                                                                                               
Aborted (core dumped)
```

& with gdb

```
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".                                                              
**************                                                                                                                             
MALOS starting                                                                                                                             
**************              
You can query specific driver info using port 20012.                                                                                       
Starting initialization of  IMU                                                                                                            
[New Thread 0x76d51440 (LWP 472)]                                                                                                          
[New Thread 0x76551440 (LWP 473)]                                                                                                          
[New Thread 0x75d51440 (LWP 474)]                                                                                                          
[New Thread 0x753ff440 (LWP 475)]                                                                                                          
[New Thread 0x74bff440 (LWP 476)]                                                                                                          
[New Thread 0x743ff440 (LWP 477)]                                                                                                          
[New Thread 0x739ff440 (LWP 478)]                                                                                                          
[New Thread 0x731ff440 (LWP 479)]                                                                                                          
[New Thread 0x729ff440 (LWP 480)]                                                                                                          
[New Thread 0x721ff440 (LWP 481)]                                                                                                          
[New Thread 0x719ff440 (LWP 482)]                                                                                                          
[New Thread 0x711ff440 (LWP 483)]                                                                                                          
Creating threads of  IMU            
Creating config thread  IMU                                                                                                                
[New Thread 0x707ff440 (LWP 484)]                                                                                                          
Creating keepalive thread  pure virtual method called                                                                                      
terminate called without an active exception                                                                                               
IMU                                                                                                                                        
                                                                                                                                           
Program received signal SIGABRT, Aborted.                                                                                                  
[Switching to Thread 0x707ff440 (LWP 484)]                                                                                                 
0x76d698e6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6                                                                                
(gdb)            
```

arhuaco, 2016-10-14:
@buildog 

I not sure about how to proceed from here :-/ But we can get feedback from more people tomorrow. Last thing I can think of is making sure that the distribution is up-to-date with:

    apt-get update && apt-get upgrade

And restart. MALOS runs as a service now so you don't need to start it by hand.

I've checked the cases where this can happen and it doesn't make sense for MALOS...

Sorry I cannot help more now, but let's see what happens tomorrow.

buildog, 2016-10-14:
yep it's up to date. 
Dont worry and thanks for your efforts @arhuaco !

buildog, 2016-10-14:
As I'm using resin.io, this dockerfile may help to reproduce the bug:

```
# base-image for node on any machine using a template variable,
# see more about dockerfile templates here: http://docs.resin.io/deployment/docker-templates/
# and about resin base images here: http://docs.resin.io/runtime/resin-base-images/
# Note the node:slim image doesn't have node-gyp
FROM resin/%%RESIN_MACHINE_NAME%%-node:slim

# use apt-get if you need to install dependencies,
RUN echo "deb http://packages.matrix.one/matrix-creator/ ./" | sudo tee --append /etc/apt/sources.list
RUN apt-get update && sudo apt-get upgrade
RUN apt-get install -y libzmq3-dev xc3sprog malos-eye matrix-creator-openocd wiringpi matrix-creator-init matrix-creator-malos cmake g++ git --force-yes;

## AUDIO
RUN apt-get install swig3.0 python-pyaudio python3-pyaudio sox
RUN apt-get -q -y install python build-essential python-dev python-pip
RUN pip install pyaudio
RUN apt-get install libatlas-base-dev

# Defines our working directory in container
WORKDIR /usr/src/app

# Copies the package.json first for better cache on later pushes
COPY package.json package.json

# This install npm dependencies on the resin.io build server,
# making sure to clean up the artifacts it creates in order to reduce the image size.
RUN JOBS=MAX npm install --production --unsafe-perm && npm cache clean && rm -rf /tmp/*

# This will copy all files in our root to the working  directory in the container
COPY . ./

# Enable systemd init system in container
ENV INITSYSTEM=on

# server.js will run when container starts up on the device
CMD ["npm", "start"]
```

arhuaco, 2016-10-14:
Oh! I was about to ask for this... (What distro you were using)

The recipe should then work with resin/rpi-raspbian, jessie tag.

This is most likely a binary compatibility issue... The packages are for an updated Raspbian Jessie and if the libc is different things won't work.

We will need a recipe to build stuff from sources, but with this route malos-eye is not available.

arhuaco, 2016-10-14:
MALOS should build though... And the rest of the programs... I wonder how much Resin changes the base images, hopefully not too much.

buildog, 2016-10-14:
I'll try with a resin/raspberrypi-debian image and let you know, aahah it's 6am :-)

buildog, 2016-10-17:
@arhuaco it worked with the `resin/rpi-raspbian` image 👍

closed 2016-10-17 by buildog