matrix-io/matrix-creator-malos #22: Camera initialization migrated to malos. Update documentation to reflect this.

brianofrokk3r, 2016-10-14:
Update installation here.
https://github.com/matrix-io/matrix-documentation

brianofrokk3r, 2016-10-14:
@arhuaco what steps change now?

arhuaco, 2016-10-14:
No longer needed (done by matrix-creator-init -- uploading package now).

    # Enable your camera on your Pi.
    sudo raspi-config;
    sudo reboot;

No longer needed, these run as services.

    # currently needs to be run on every restart
    sudo modprobe bcm2835-v4l2;
    # before you run
    malos_eye & malos 2>&1

I can send PR.

brianofrokk3r, 2016-10-14:
@arhuaco perfect.

arhuaco, 2016-10-14:
I couldn't create a branch in the doc repo. Here is a patch in the mean time.

[config-update.patch.gz](https://github.com/matrix-io/matrix-creator-malos/files/530155/config-update.patch.gz)

arhuaco, 2016-10-20:
This is pending. The previous patch is easy to apply... Pinging so that we can close this issue.

brianofrokk3r, 2016-10-20:
@arhuaco this is done

closed 2016-10-20 by arhuaco