matrix-io/matrix-creator-malos #23: alsa capture devices

buildog, 2016-10-17:
I'm trying to test the micarray demo with alsa, but it's look like alsa can't find any capture device : 

```
root@raspberrypi3-82d23ee:/usr/src/app# arecord -l                                                                                                 
**** List of CAPTURE Hardware Devices ****  
```

Did I missed something ?

brianofrokk3r, 2016-10-17:
@eljuguetero

buildog, 2016-10-18:
Look like same issue there : https://github.com/alexylem/jarvis/issues/103#issuecomment-253940683

arhuaco, 2016-10-19:
@buildog Can you share the recipe you're using? I can try replicating with resin.

arhuaco, 2016-10-20:
Hi @buildog, I tried to reproduce but things are working for me.

The steps:

1) Download Raspbian Jessie Lite and flash it into the micro-sd.
2) Followed the Getting started steps ( https://matrix-io.github.io/matrix-documentation/#raspberry-pi-setup ). It's no longer necessary to enable the camera by hand, it's done by matrix-creator-init.
3) reboot

On boot I find this:

    $ ls /tmp/
    matrix_micarray_channel_0  matrix_micarray_channel_1  matrix_micarray_channel_2 matrix_micarray_channel_3  matrix_micarray_channel_4  matrix_micarray_channel_5 matrix_micarray_channel_6  matrix_micarray_channel_7  sam3-program.log

    $ arecord -l
    **** List of CAPTURE Hardware Devices ****
    card 1: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
      Subdevices: 8/8
      Subdevice #0: subdevice #0
      Subdevice #1: subdevice #1
      Subdevice #2: subdevice #2
      Subdevice #3: subdevice #3
      Subdevice #4: subdevice #4
      Subdevice #5: subdevice #5
      Subdevice #6: subdevice #6
      Subdevice #7: subdevice #7 

Then I tried the MIC:

    arecord -d 10 --device=mic_channel0 -r 16000 -c 1  -f S16_LE test.wav

And I could play the audio in my computer.

What could be different in your setup?

buildog, 2016-10-23:
hey @arhuaco, sorry for the delay

```
root@raspberrypi3-82d23ee:/usr/src/app# ls /tmp/                                                                                                   
matrix_micarray_channel_0  matrix_micarray_channel_2  matrix_micarray_channel_4  matrix_micarray_channel_6  sam3-program.log                       
matrix_micarray_channel_1  matrix_micarray_channel_3  matrix_micarray_channel_5  matrix_micarray_channel_7                                         
root@raspberrypi3-82d23ee:/usr/src/app#  arecord -l                                                                                                
**** List of CAPTURE Hardware Devices ****    
```

from this dockerconf (rpi-raspbian+node+matrix+electron+snowboy)

```
FROM resin/rpi-raspbian

# Install wget and curl
RUN apt-get update && apt-get install -y \
  wget \
  curl && rm -rf /var/lib/apt/lists/*

# Install other apt deps
RUN apt-get update && apt-get install -y \
  apt-utils \
  build-essential \
  clang \
  xserver-xorg-core \
  xserver-xorg-input-all \
  xserver-xorg-video-fbdev \
  xorg \
  libdbus-1-dev \
  libgtk2.0-dev \
  libnotify-dev \
  libgnome-keyring-dev \
  libgconf2-dev \
  libasound2-dev \
  libcap-dev \
  libcups2-dev \
  libxtst-dev \
  libxss1 \
  libnss3-dev \
  fluxbox \
  libsmbclient \
  libssh-4 \
  fbset \
  libexpat-dev && rm -rf /var/lib/apt/lists/*

# Set Xorg and FLUXBOX preferences
RUN mkdir ~/.fluxbox
RUN echo "xset s off\nxserver-command=X -s 0 dpms" > ~/.fluxbox/startup
RUN echo '#!/bin/sh\n\nexec /usr/bin/X -s 0 dpms -nocursor -nolisten tcp "$@"' > /etc/X11/xinit/xserverrc

RUN echo "deb http://packages.matrix.one/matrix-creator/ ./" | sudo tee --append /etc/apt/sources.list
RUN apt-get update && sudo apt-get upgrade

# MATRIX
RUN apt-get -q -y install python build-essential python-dev python-pip
RUN apt-get install -y libzmq3-dev xc3sprog malos-eye matrix-creator-openocd wiringpi matrix-creator-init matrix-creator-malos cmake g++ git --force-yes;

#### NODE
RUN sudo apt-get install curl
RUN curl -sL https://deb.nodesource.com/setup_6.x | sudo bash -
RUN sudo apt-get install nodejs
RUN apt-get install make

## AUDIO (snowboy)
RUN apt-get install swig3.0 python-pyaudio python3-pyaudio sox
RUN pip install pyaudio
RUN apt-get install libatlas-base-dev
RUN sudo apt-get install alsa-base alsa-utils

# Save source folder
RUN printf "%s\n" "${PWD##}" > SOURCEFOLDER

# Move to app dir
WORKDIR /usr/src/app

# Move package.json to filesystem
COPY package.json package.json

# NPM i app
RUN sudo npm i --production

# NPM rebuild node native modules after electron is installed.
RUN sudo apt-get install libssl-dev
RUN sudo ./node_modules/.bin/electron-rebuild --pre-gyp-fix

RUN sudo npm rebuild zmq

# This will copy all files in our root to the working  directory in the container
COPY . ./

## uncomment if you want systemd
ENV INITSYSTEM on

# Start electron
CMD ["npm", "run", "start"]
```

LengZai, 2016-10-23:
Hi @arhuaco 

Well, i needed to prepare a new microSD to test but i have tried multiple things today but i didn't find any solution and i'm still having this useless error message:
**arecord: main:722: audio open error: No such file or directory**
What a pity!
Waiting @buildog tries from matrix-io github to know if he found a solution

Can you post the output of this command (uppercase L instead of lowercase l):
$>arecord -L

Thanks.

In the meantime, I am going to find a microSD to have a fresh Raspberry Jessie Lite (as i'm using a no-Lite version)

buildog, 2016-10-25:
```
root@raspberrypi3-82d23ee:/usr/src/app# arecord -L                                                                                                 
null                                                                                                                                               
    Discard all samples (playback) or generate zero samples (capture)                                                                              
mic_channel0                                                                                                                                       
mic_channel1                                                                                                                                       
mic_channel2                                                                                                                                       
mic_channel3                                                                                                                                       
mic_channel4                                                                                                                                       
mic_channel5                                                                                                                                       
mic_channel6                                                                                                                                       
mic_channel7              
``

```
root@raspberrypi3-82d23ee:/usr/src/app# arecord -d 10 --device=mic_channel0 -r 16000 -c 1  -f S16_LE test.wav                                      
arecord: main:722: audio open error: No such file or directory  
```

LengZai, 2016-11-01:
Hi @arhuaco

Can you post the output of this command (uppercase L instead of lowercase l):
$>arecord -L

Thanks

arhuaco, 2016-11-01:
@LengZai Here it is:

```
arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
mic_channel0
mic_channel1
mic_channel2
mic_channel3
mic_channel4
mic_channel5
mic_channel6
mic_channel7
default:CARD=Dummy
    Dummy, Dummy PCM
    Default Audio Device
sysdefault:CARD=Dummy
    Dummy, Dummy PCM
    Default Audio Device
dmix:CARD=Dummy,DEV=0
    Dummy, Dummy PCM
    Direct sample mixing device
dsnoop:CARD=Dummy,DEV=0
    Dummy, Dummy PCM
    Direct sample snooping device
hw:CARD=Dummy,DEV=0
    Dummy, Dummy PCM
    Direct hardware device without any conversions
plughw:CARD=Dummy,DEV=0
    Dummy, Dummy PCM
    Hardware device with all software conversions
```

buildog, 2016-11-08:
@arhuaco still the same error there, as I have some time to spend I'm trying to figure what's wrong.

Could you please give me what's in your /etc/.asoundrc file ?

arhuaco, 2016-11-08:
It's here: https://github.com/matrix-io/matrix-creator-init/blob/master/asound.conf

arhuaco, 2016-11-08:
Not that one. /etc/.asoundrc doesn't exist...

buildog, 2016-11-08:
yes .asoud.conf, I've the same.

I tried with an USB mic and it worked :-/

arhuaco, 2016-11-08:
Can you try Raspbian? Just to see what happens.

buildog, 2016-11-08:
mhh I think @LengZai is using Raspbian and has the same issue https://github.com/alexylem/jarvis/issues/103#issuecomment-253940683

arhuaco, 2016-11-08:
If you can please try the following instructions. If the basic instructions work you can compare running environments. So far we've only had reports from you both, and you've followed advanced paths. Not that it's wrong, but many things can fail in environments that are different.

https://github.com/matrix-io/matrix-creator-quickstart/wiki/2.-Getting-Started

LengZai, 2016-11-08:
Hi

Here mine:

> $ arecord -L
> null
>     Discard all samples (playback) or generate zero samples (capture)
> pulse
>     PulseAudio Sound Server
> mic_channel0
> mic_channel1
> mic_channel2
> mic_channel3
> mic_channel4
> mic_channel5
> mic_channel6
> mic_channel7
> sysdefault:CARD=Dummy
>     Dummy, Dummy PCM
>     Default Audio Device
> dmix:CARD=Dummy,DEV=0
>     Dummy, Dummy PCM
>     Direct sample mixing device
> dsnoop:CARD=Dummy,DEV=0
>     Dummy, Dummy PCM
>     Direct sample snooping device
> hw:CARD=Dummy,DEV=0
>     Dummy, Dummy PCM
>     Direct hardware device without any conversions
> plughw:CARD=Dummy,DEV=0
>     Dummy, Dummy PCM
>     Hardware device with all software conversions
> 

So I suspect the PulseAudio and

>  default:CARD=Dummy
>     Dummy, Dummy PCM
>     Default Audio Device

Thanks to both... Still investigating before trying to reinstall Jessie :-(
Note: I'm using Raspbian Jessie with last update (I have just done apt-get update and upgrade)
I didn't try with Raspbian Jessie Lite one but i don't think that this issue is related between Lite and no-Lite.

LengZai, 2016-11-09:
## **Finally !**

Thanks a lot @arhuaco !
I succeed in making work! but I don't really understand how to let Raspberry to select the default audio device and the capture audio device... but everything is related to these 3 files:

> ~/.asound.rc
> /etc/modules-load.d/asound.conf
> /etc/modprobe.d/alsa-base.conf

@arhuaco 
Need some details to confirm how i make it work... How you "test" the matrix creator with Pi..

- Raspbian Jessie version? Lite or not?
- Are you directly "su" on your pi or you use a standard user?
- Can you give me the output of this command: 

> cat /proc/asound/modules

I found that the file asound.conf has been created in /etc/asound.conf on October 4th. Date when i tried the Matrix when you release alsa support
Wondering if this file has been put on the right place?
I didn't touch this file on asound.conf but looks like the one on your link.. so i have created the same file in /etc/modules-load.d/
But i didn't see any difference except if i launch the arecord with sudo. (But i didn't check if the sound is recorded (as i don't get sound through aplay)

So to make it work... I copy the content of asound.conf in .asoundrc (=> which located on my user home folder ~/). Note: I use my default standard user "pi" default user name from default Raspberry Jessie installation

FYI: here my /proc/asound/modules output:

> $ cat /proc/asound/modules
>  0 snd_dummy
>  2 snd_bcm2835

Because i added the "dummy driver" in alsa-base.conf file

> cat /etc/modprobe.d/alsa-base.conf
> \# This sets the index value of the cards but doesn't reorder.
> options snd_dummy index=0
> options snd_usb_audio index=1
> options snd_bcm2835 index=2
> 
> \# Does the reordering.
> options snd slots=snd_dummy,snd_usb_audio,snd_bcm2835
> 

(I didn't plug any usb audio device at that time, that's why you don't see 1 snd_usb_audio from cat /proc/asound/modules)

With this command line... i can record my voice now (so the one that you provide @arhuaco 
arecord -d 10 --device=mic_channel0 -r 16000 -c 1 -f S16_LE test.wav

I suppose i need to put -d 10 instead of -d 00 because you put "hw 1" in the asound.conf (so in my .asoundrc file) but on my alsa-base.conf i put index to 0

All these tests, let me think that asound.conf is for root user (su) or at least user who has the permission "to use" it
And the .asoundrc is per user account "derivated" conf.
When i delete .asoundrc, it automatically reappears on reboot =/

Ok so now my question is... how to make a clean and proper .asoundrc (or asound.conf whatever) to let me do something like this:

> arecord -r 16000 -c 1 -f S16_LE test.wav
> aplay test.wav

arecord uses the mic_array
aplay uses the default Raspberry 3.5mm jack output

So to find the proper asound config to make all by default.
Hope your answers of my questions above will help us to figure out that... so i will be able to use jarvis project with Matrix Creator

last comment: My voice was quite low even i put high volume from alsamixer for the capture device and play device

Feel feel to ask me any questions on my post.. Bit late on my side so i didn't read back my message :P

arhuaco, 2016-11-10:
Hi there.

- Raspbian Lite.
- Non-root user.

```
cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_dummy
```

The package matrix-creator-init places creator-mics.conf in /etc/modules-load.d .

I also get low volume. We might be able to increase the gain. I will update this thread soon.

LengZai, 2016-11-10:
Thanks... so by default, you have snd_dummy "activated"
That's why it works on your side.

Can you give me your output for this file please?
`cat /etc/modprobe.d/alsa-base.conf`

arhuaco, 2016-11-10:
I don't have the file. These are the files that match the word "alsa":

    find /etc -iname "*alsa*"
    /etc/init.d/alsa-utils
    /etc/rcS.d/S16alsa-utils
    /etc/rc6.d/K01alsa-utils
    /etc/rc0.d/K01alsa-utils
    /etc/rc1.d/K01alsa-utils

Attaching [alsa-utils.gz](https://github.com/matrix-io/matrix-creator-malos/files/584366/alsa-utils.gz) in case it helps.

LengZai, 2016-11-10:
Hi .. here some progress @buildog 

Put this on your ~/.asoundrc
Change card and device number based on your pi configuration. Read my comments on my .asoundrc file
I think by playing with dsnoop from alsa option, it's possible to "merge" the 8 mics together but i'm still investigating on that... [http://alsa.opensrc.org/Dsnoop](http://alsa.opensrc.org/Dsnoop)

With this (user) configuration file, you will be able to use arecord and aplay without specifying any device on parameter:
So now i can use arecord with Matrix creator mic_array (mic_channel0) with this command
`$> arecord -r 16000 -c 1 -f S16_LE test.wav`
And play with aplay on the Pi3 jack output with this command:
`$>aplay test.wav`
Hope it will help the Matrix community

```
#To select the default sound card. To get the card/device combinaison list $>aplay -l
#
pcm.!default {
    type hw
    card 2
    device 0
}

#To select the default sound card for controlling the volume. Should be the same like pcm. To get the list $>aplay -l
#
ctl.!default {
    type hw
    card 2
    device 0
}

#Set Default playback and capture card. I use Asym because the playback and capture devices are not on the same device.
#So it will merge the 2 card like having an unique full-duplex sound card
#
pcm.!default
{
    type asym
    playback.pcm
    {
        type plug
        slave.pcm "hw:2,0"
    }
    capture.pcm
    {
        type plug
        slave.pcm "mic_channel0"
    }
}

#For get Matrix Creator mic array working with ALSA. To get the card/device combinaison list $>arecord -l
#You should find Dummy card. Check that you have snd_dummy enabled $>cat /proc/asound/modules
#If you don't find snd_dummy, you have to change your /etc/modprobe.d/alsa-base.conf to add the following lines:
#
# $>cat /etc/modprobe.d/alsa-base.conf
# #This sets the index value of the cards but doesn't reorder.
# options snd_dummy index=0
# options snd_usb_audio index=1
# options snd_bcm2835 index=2
#
# #Does the reordering.
# options snd slots=snd_dummy,snd_usb_audio,snd_bcm2835
#
pcm.sc {
    type hw
    card 0
    device 0
}

pcm.mic_channel0 {
  type file
  file "/tmp/matrix_micarray_channel_0"
  infile "/tmp/matrix_micarray_channel_0"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel1 {
  type file
  file "/tmp/matrix_micarray_channel_1"
  infile "/tmp/matrix_micarray_channel_1"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel2 {
  type file
  file "/tmp/matrix_micarray_channel_2"
  infile "/tmp/matrix_micarray_channel_2"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel3 {
  type file
  file "/tmp/matrix_micarray_channel_3"
  infile "/tmp/matrix_micarray_channel_3"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel4 {
  type file
  file "/tmp/matrix_micarray_channel_4"
  infile "/tmp/matrix_micarray_channel_4"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel5 {
  type file
  file "/tmp/matrix_micarray_channel_5"
  infile "/tmp/matrix_micarray_channel_5"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel6 {
  type file
  file "/tmp/matrix_micarray_channel_6"
  infile "/tmp/matrix_micarray_channel_6"
  format "raw"
  slave {
    pcm sc
  }
}

pcm.mic_channel7 {
  type file
  file "/tmp/matrix_micarray_channel_7"
  infile "/tmp/matrix_micarray_channel_7"
  format "raw"
  slave {
    pcm sc
  }
}

```

LengZai, 2016-11-10:
Hi @arhuaco 

It seems that Matrix Creator can only record at 16000Hz rate?
When i try
`$>arecord -r 48000 -c 1 -f S16_LE test.wav`
I cannot play it correctly... (super fast)
In 16000Hz i dont have any prob
Possible to fix that?

buildog, 2016-11-11:
Hey @LengZai great to see some change there, and yeah merging the 8 mic is a must have, I totaly agree.

As @arhuaco I don't have the ` /etc/modprobe.d/alsa-base.conf` too, look like it moved to `/usr/share/alsa/alsa.conf` on my disto , but the key there is that we don't have the snd-dummy driver loaded at boot + the wrong default card number (at least a wrong config file loaded depending of the user .asoundrc or globaly asound.conf)

Here is how I finaly make it record: 

**Load driver** 

```
sudo modprobe snd-dummy
cat /proc/asound/modules                                                                                   
 0 snd_bcm2835                                                                                                                                     
 1 snd_dummy          
```

**Change defaults**

In `usr/share/alsa/alsa.conf` to set differents defaults to 1 (snd_dummy)

```
defaults.ctl.card 1
defaults.pcm.card 1
```

**Record!**

```
arecord -r 16000 -c 1 -f S16_LE test.wav                                                                   
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono  
```

**Consistant fix**

Load the driver at boot in the right order (snd-dummy then bcm), I guess this should fix the issue without having to change too much the configuration... still have to try it

```
sudo nano /etc/modules
```

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-dummy
snd-bcm2835
```

arhuaco, 2016-11-11:
> and yeah merging the 8 mic is a must have, I totaly agree.

This should be added next week. As an additional mic.

buildog, 2016-11-12:
Great so I can close this one

closed 2016-11-12 by buildog

samwel89, 2017-03-29:
Hi all, 

I'm interested in the subject of the use of 8 microarrays on jarvis. I have recently bought a matrix creator and I want to use it with jarvis. I have well installed the matrix OS and I have no problem so far, However, I have not yet succeeded to trigger a hotword to wake-up jarvis. I have tested this command ***arecord --device=mic_channel1 -r 16000 -c 1 -f S16_LE test.wav* but it get timed out without hearing the sound.

However, I have played **aplay channel_0.wav** and I can hear the sound though there are some noises.
I was wondering, if @LengZai got it worked, or any progress from the Matrix team? Any help or suggestions will be much appreciated.
Thank you.

Samuel.