matrix-io/matrix-creator-malos #33: node test_everloop.js ->TypeError: Cannot read property 'build' of null

adrian2lee, 2016-10-26:
Getting following error code
> /home/pi/matrix-creator-malos/src/js_test/test_everloop.js:17
> var matrixMalosBuilder = protoBuilder.build("matrix_malos")
>                                      ^
> 
> TypeError: Cannot read property 'build' of null
>     at Object.<anonymous> (/home/pi/matrix-creator-malos/src/js_test/test_everloop.js:17:38)
>     at Module._compile (module.js:556:32)
>     at Object.Module._extensions..js (module.js:565:10)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)
>     at Function.Module._load (module.js:424:3)
>     at Module.runMain (module.js:590:10)
>     at run (bootstrap_node.js:394:7)
>     at startup (bootstrap_node.js:149:9)
>     at bootstrap_node.js:509:3

brianofrokk3r, 2016-10-26:
Hey @adrian2lee, try running the following in the cloned directory.
```
git submodule update --init
```

Then run `npm install` in the `src/js_test` directory. Retry the same script and let me know.

adrian2lee, 2016-10-26:
@brianofrokk3r thanks, it works!

closed 2016-10-26 by adrian2lee

brianofrokk3r, 2016-10-26:
@adrian2lee which docs did you follow to set up? I want to make sure we update it.

adrian2lee, 2016-10-26:
i tried both 
https://matrix-io.github.io/matrix-documentation/MALOS/example/
and
https://github.com/matrix-io/matrix-creator-quickstart/wiki/4.-MALOS

luisnomad, 2016-12-03:
@adrian2lee running 

`git submodule update --init`

and then `npm install` in the `src/js_test` did not fix it for me.

I get this:

```
pi@raspberrypi:~/matrix-creator-malos/src/js_test $ node test_humidity.js 
/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:22
var protoBuilder = protoBuf.loadProtoFile('../../protocol-buffers/malos/driver.proto')
                            ^

TypeError: protoBuf.loadProtoFile is not a function
    at Object.<anonymous> (/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:22:29)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
```

ada2018, 2017-04-06:
still can see this error when using Electron

eighteyes, 2017-04-06:
Can you give us steps to reproduce?

ada2018, 2017-04-06:
It works now, Thanks for the reply, error is fixed by specifying the right  relative path for proto file, thanks