matrix-io/matrix-creator-malos #36: IMU Noise

nebiljabari, 2016-10-29:
On a static plan (a table) during 30s, with 5s pings,  I have a range fluctuation order of :

yaw   :  min 151.5074462890625, max 152.98768615722656 (difference : 1.4802398681640625)
pitch : min 1.4778050184249878, max 1.7689162492752075 (diff : 0.2911112308502197)
roll    : min -0.529462456703186, max -0.31852051615715027  (diff 0.21094194054603577)

Can it be considered as "acceptable noise" for thin project like drone ?
If no, will you improve the robustness of data reading with future updates ?

brianofrokk3r, 2016-11-01:
@hpsaturn @eljuguetero @yoelrc88

nebiljabari, 2016-11-14:
guys @hpsaturn @eljuguetero @yoelrc88 some info or feedback from experience would be appreciated

yoelrc88, 2016-11-14:
Hi  @nebiljabari , 

Interesting question. To truly characterize the quality of the signal is better to have a smaller sample time than 5s, at least in ms, specially for projects like drone that require frequent updates in the control loop. Also max & min  does give you to much info about what is going on there, plotting tools and FFT analysis can give you a better idea of the signal. 
About the errors you show I think the pitch and roll are pretty decent for any application. The yaw value seems a little more unstable (not bad). Take into account that for drones fly control the PITCH and ROLL errors are more critical than the YAW, is what keeps the drone stable , the YAW is mostly for heading.

Take into account that a drone project is always a challenge y terms of sampling time, control, tuning, etc. You will need to go down closer to the hardware to implement a flight controller for a drone. Maybe using just HAL or even making your own firmware for the MCU or the FPGA. For other applications like orientation detection, robotics, movements detection is going to be easier.

Hope this help, 

Yoel

nebiljabari, 2016-11-17:
Thanks @yoelrc88 for your detailed response. Will follow your advice

closed 2016-11-17 by nebiljabari