matrix-io/matrix-creator-malos #48: Could not parse protobuf.

saranvs, 2016-11-25:
Hi All,
           I got an error while running "test_humidity.js" , My Ip address is 172.17.8.39:20017 when i give a request it throw the following error " Message received: Humidity error: 0, Invalid configuration for Humidity driver. Could not parse protobuf." Can anyone help me to resolve this issue

{ humidity: 33.425506591796875,
  temperature: 23.233409881591797,
  temperature_raw: 38.9051628112793,
  temperature_is_calibrated: true }
{ humidity: 33.390079498291016,
  temperature: 23.186725616455078,
  temperature_raw: 38.886451721191406,
  temperature_is_calibrated: true }
Message received: Humidity error: 0, Invalid configuration for Humidity driver. Could not parse protobuf.
Message received: Humidity error: 0, Invalid configuration for Humidity driver. Could not parse protobuf.
{ humidity: 33.30833053588867,
  temperature: 23.233409881591797,
  temperature_raw: 38.9051628112793,
  temperature_is_calibrated: true }
{ humidity: 33.19660568237305,
  temperature: 24.084392547607422,
  temperature_raw: 38.92387008666992,
  temperature_is_calibrated: true }
{ humidity: 33.82607650756836,

hpsaturn, 2016-11-25:
Maybe you have old version of matrix-creator-malos deb package. Please on your rasberry pi get the package version: (matrix-creator-malos 0.1-4.8)
```bash
sudo dpkg -l | grep matrix-creator-malos
``` 
and please verify the last version on repository (master) with ```git pull```

saranvs, 2016-11-26:
Hi hpsaturn, 
   I checked the version it is **Version: 0.1-4.8**.

saranvs, 2016-11-26:
I have done **git pull** and/or **git submodule**  inside **matrix-creator-malos** folder ,but still facing the same issue.

For your Understanding..
pi@raspberrypi:~/matrix-creator-malos $ git pull
remote: Counting objects: 81, done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 81 (delta 50), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (81/81), done.
From https://github.com/matrix-io/matrix-creator-malos
   2313c57..ec15b91  master     -> origin/master
 * [new branch]      av/doc_gpio -> origin/av/doc_gpio
 * [new branch]      av/imu_missing_vars -> origin/av/imu_missing_vars
   530fe80..17549c9  av/servo_driver -> origin/av/servo_driver
   b62c5ff..5725cd9  kp/doc     -> origin/kp/doc
Fetching submodule protocol-buffers
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 17 (delta 6), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (17/17), done.
From https://github.com/matrix-io/protocol-buffers
 * [new branch]      av/imu_missing_vars -> origin/av/imu_missing_vars
   50a780b..e1b9e61  master     -> origin/master
 * [new branch]      yc/fix_temp_calib -> origin/yc/fix_temp_calib
Updating 2313c57..ec15b91
error: Your local changes to the following files would be overwritten by merge:
        src/js_test/test_humidity.js
Please, commit your changes or stash them before you can merge.
Aborting
pi@raspberrypi:~/matrix-creator-malos $ git submodule
 50a780b70bd1d545c98fed1b21c06faeddad1038 protocol-buffers (heads/master)

arhuaco, 2016-11-26:
Hi, a fix was submitted yesterday. Sorry for the inconvenience.

https://github.com/matrix-io/protocol-buffers/commit/e1b9e61d193e4f0fdafecae6ef6bb55244a85d52

I just updated matrix-creator-malos to point to the latest protobufs.

saranvs, 2016-11-28:
Hi,
         Sorry for the late reply. Still facing the same issue.

arhuaco, 2016-11-28:
I tried:

```
$ git clone https://github.com/matrix-io/matrix-creator-malos.git
$ cd matrix-creator-malos/src/js_test/
$ npm install
$ git submodule update --init
```

Now I get this:

```
$ node test_humidity.js 
Sending pings every 5 secondsMessage received: Humidity error: 0, Invalid specific configuration for Humidity driver.
{ humidity: 74.29014587402344,
  temperature: 35.503807067871094,
  temperature_raw: 35.503807067871094,
  temperature_is_calibrated: false }
{ humidity: 74.3109359741211,
  temperature: 35.50358581542969,
  temperature_raw: 35.50358581542969,
  temperature_is_calibrated: false }
{ humidity: 74.31761932373047,
  temperature: 35.501792907714844,
  temperature_raw: 35.501792907714844,
  temperature_is_calibrated: false }
{ humidity: 74.3346939086914,
  temperature: 35.502464294433594,
  temperature_raw: 35.502464294433594,
  temperature_is_calibrated: false }
{ humidity: 74.3346939086914,
  temperature: 35.502464294433594,
  temperature_raw: 35.502464294433594,
  temperature_is_calibrated: false }
{ humidity: 74.29905700683594,
  temperature: 35.50358581542969,
  temperature_raw: 35.50358581542969,
  temperature_is_calibrated: false }
```

I get an error but it's not about the protobuf. I think @yoelrc88 can help us here because he was the last one to update this program.

You probably can switch to the latest protos (without doing all the steps above) doing this:

```
$ cd protocol-buffers
$ git checkout master
$ git pull
```

But if it doesn't work it's better to do the whole thing.

saranvs, 2016-11-28:
Hi,

 This was what i got

pi@raspberrypi:~/matrix-creator-malos/src/js_test $ node test_humidity.js
Sending pings every 5 secondsMessage received: Humidity error: 0, Invalid specif                                                                                                                                                             ic configuration for Humidity driver.
{ humidity: 33.834251403808594,
  temperature: 39.952903747558594,
  temperature_raw: 39.952903747558594,
  temperature_is_calibrated: false }
{ humidity: 33.5372314453125,
  temperature: 39.89677429199219,
  temperature_raw: 39.89677429199219,
  temperature_is_calibrated: false }
{ humidity: 34.338375091552734,
  temperature: 39.952903747558594,
  temperature_raw: 39.952903747558594,
  temperature_is_calibrated: false }
{ humidity: 34.338375091552734,
  temperature: 39.952903747558594,
  temperature_raw: 39.952903747558594,
  temperature_is_calibrated: false }
{ humidity: 33.90237808227539,
  temperature: 40.009033203125,
  temperature_raw: 40.009033203125,
  temperature_is_calibrated: false }
{ humidity: 33.526329040527344,
  temperature: 39.8780632019043,
  temperature_raw: 39.8780632019043,
  temperature_is_calibrated: false }
{ humidity: 33.82062911987305,
  temperature: 39.9341926574707,
  temperature_raw: 39.9341926574707,
  temperature_is_calibrated: false }
{ humidity: 33.87785339355469,
  temperature: 40.027740478515625,
  temperature_raw: 40.027740478515625,
  temperature_is_calibrated: false }
{ humidity: 33.87785339355469,
  temperature: 40.027740478515625,
  temperature_raw: 40.027740478515625,
  temperature_is_calibrated: false }
{ humidity: 33.796104431152344,
  temperature: 39.97161102294922,
  temperature_raw: 39.97161102294922,
  temperature_is_calibrated: false }
Message received: Humidity error: 0, Invalid configuration for Humidity driver. Could not parse protobuf.
Message received: Humidity error: 0, Invalid configuration for Humidity driver. Could not parse protobuf.
{ humidity: 33.86150360107422,
  temperature: 39.84064483642578,
  temperature_raw: 39.84064483642578,
  temperature_is_calibrated: false }
{ humidity: 33.80427932739258,
  temperature: 39.97161102294922,
  temperature_raw: 39.97161102294922,
  temperature_is_calibrated: false }
{ humidity: 33.80427932739258,
  temperature: 39.97161102294922,
  temperature_raw: 39.97161102294922,
  temperature_is_calibrated: false }
{ humidity: 33.82880401611328,
  temperature: 39.91548538208008,

arhuaco, 2016-11-28:
It seems the problem is in the MALOS driver for humidity configuration. An error similar to the one I saw.

In the mean time you can ignore the error since you can get the values. We'll check this soon.

BTW, did you modify the example?

saranvs, 2016-11-28:
I have only modified  the IP address from 127.0.0.1 to 172.17.8.39

arhuaco, 2016-11-28:
Ok. Thanks for the report. We'll take a look soon.

saranvs, 2016-11-28:
Hi ,
    *     I have run a set of examples such as  1) test_humidity.js   2)  query_demographics.js ( maloe-eye)       3) test_everloop.js  4) test_pressure.js,  in all those examples I got the same error "Could not parse protobuf".

    *  Alexa Voice demo works perfectly.

brianofrokk3r, 2016-11-28:
Hey @saranvs,

Can you `cd` into the `proto` directory in the `matrix-creator-malos` root directory and check what the commit number is? 

For reference, the latest version is `e1b9e61d193e4f0fdafecae6ef6bb55244a85d52`.
https://github.com/matrix-io/protocol-buffers/commit/e1b9e61d193e4f0fdafecae6ef6bb55244a85d52

saranvs, 2016-11-28:
Hi @brianofrokk3r ,

  Do u mean **protocol-buffers** directory  or   **proto** directory inside **matrix-creator-malos**  root directory. There is no directory named **proto**. Also can i know how to check the commit number

brianofrokk3r, 2016-12-03:
@saranvs Yes correct.

saranvs, 2016-12-07:
Hey @brianofrokk3r ,

 It's showing the latest  version.

saranvs, 2016-12-19:
HI All,

I removed all  packages for matrix creator-malos, zmq,protobuf,protobufjs, and reinstalled those thing but still i am getting the same error. Still I am the only one facing the issue.
Installed Version of Packages:
1. npm - 3.10.3
2. protobuf  - libprotoc 2.6.1
3. node - 6.5.0
4. zmq - 2.15.3
5. protobufjs -6.2.1

closed 2016-12-22 by saranvs