matrix-io/matrix-creator-malos #53: TypeError: protoBuf.loadProtoFile is not a function

luisnomad, 2016-12-03:
I can't make any example work.

I have followed the instructions suggested here https://github.com/matrix-io/matrix-creator-malos/issues/33, like running 

`git submodule update --init`

and then `npm install` in the `src/js_test` folder, but it did not fix the issues for me.

I get this:

```
pi@raspberrypi:~/matrix-creator-malos/src/js_test $ node test_humidity.js 
/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:22
var protoBuilder = protoBuf.loadProtoFile('../../protocol-buffers/malos/driver.proto')
                            ^

TypeError: protoBuf.loadProtoFile is not a function
    at Object.<anonymous> (/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:22:29)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
```

luisnomad, 2016-12-03:
I have run

node

then 

```
> var protoBuf = require("protobufjs")
> protoBuf.load
[Function: load]
> protoBuf.loadProtoFile
undefined
```

could it be that I need a specific version of protobufjs?

luisnomad, 2016-12-03:
If I change `loadProtoFile` by `load`,  then I get:

```
/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:24
var matrixMalosBuilder = protoBuilder.build("matrix_malos")
                                      ^

TypeError: protoBuilder.build is not a function
    at Object.<anonymous> (/home/pi/matrix-creator-malos/src/js_test/test_humidity.js:24:39)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
```

brianofrokk3r, 2016-12-03:
Guys, found it, the latest version `protobufjs` has problems. 

The fix should be:
```
cd ~/matrix-creator-malos/src/js_test
npm install protobufjs@5.0.1 --save
node test_everloop.js
```

luisnomad, 2016-12-03:
Thanks @brianofrokk3r, that fixes the issue!

arhuaco, 2016-12-14:
I ran into the same issue. @brianofrokk3r 's workaround helped. 

I'm reopening the issue so that more people can see the workaround and because we haven't fixed it.

Two possible fixes:

1) Look into the API changes for v6.0 and update our examples.

2) Peg protbufjs version to the one that works.

1 seems better for the long term.

brianofrokk3r, 2017-01-31:
@arhuaco resolved?

arhuaco, 2017-01-31:
npm installs 5.0.2 so this is resolved.

closed 2017-01-31 by arhuaco