matrix-io/matrix-creator-malos #58: malos does not return accelerometer data when running malos demos.

brianofrokk3r, 2017-01-26:
There's an issue when running `src/js_test` demos. I believe the issue is with MALOS because the issue is across different services. Currently receiving zeros from all accelerometer data.

arhuaco, 2017-01-26:
Can't reproduce...

```
{ yaw: -131.04244995117188,
  pitch: 1.2205123901367188,
  roll: -3.202528953552246 }
{ yaw: -135.08694458007812,
  pitch: 1.1436519622802734,
  roll: -3.117536783218384 }
```

What could be causing this?

arhuaco, 2017-01-26:
I could reproduce. This is how the data is coming from HAL, @eljuguetero, @kdpatino .

brianofrokk3r, 2017-01-26:
```
{ 
  yaw: 158.4182891845703,
  pitch: 1.6816062927246094,
  roll: -1.2669049501419067,
  accel_x: 0,
  accel_y: 0,
  accel_z: 0,
  gyro_x: 0,
  gyro_y: 0,
  gyro_z: 0,
  mag_x: 0,
  mag_y: 0,
  mag_z: 0,
  type: 'gyroscope' 
}
```

eljuguetero, 2017-01-27:
Hi,
Please test
`sudo apt update &&  sudo apt upgrade
halt
`
then power off/power on the raspberry

BR,

- andres

brianofrokk3r, 2017-01-27:
@eljuguetero works! I just did `sudo shutdown -r now` and it was okay.

closed 2017-01-27 by brianofrokk3r

sunwell1994, 2020-05-22:
why gyro's value is so huge if it is static?