matrix-io/matrix-creator-malos #74: nothing appear when use node to test every sensor

arleneF, 2017-05-17:
Hi,
I followed the every step here: https://github.com/matrix-io/matrix-creator-malos, and in the **Examples**, I use `git clone https://github.com/matrix-io/matrix-creator-malos.git && cd matrix-creator-malos` to download malos, and in` js_test` folder, I test all the sensor using node, but all of them shows `Sending pings every 5 seconds` and then nothing appears.

I already did update and upgrade, also I check `ps aux | grep "malos"` , then kill and re-run it. But still failed. There is one thing I'm thinking about is that after I kill the malos and run malos again, I need to use Ctrl-C to get out, otherwise I cannot do any further work. Then I check `ps aux | grep "malos"` again, and there is only one malos:`pi        4417  0.0  0.2   4276  2016 pts/0    S+   16:09   0:00 grep --color=auto malos
` is that the problem?
If not, how can I fix it?

arleneF, 2017-05-17:
I fixed it now. I need to run malos or malos_eye in one terminal without using Ctrl-C, and then use another terminal to test the code. 
But, is there any way that we can run malos and malos_eye in background rather than use second terminal prompt?

gordinh, 2017-05-18:
After the matrix installation https://matrix-io.github.io/matrix-documentation/, MALOS and MALOS_EYE run by default in the back ground after boot.

If you want to stop them and start them again in background use:

sudo malos &
sudo malos_eye

closed 2017-05-18 by arleneF