matrix-io/matrix-creator-malos #76: Python Examples are broken

erinxocon, 2017-05-31:
It seems as if the protobuffers had changed a bit and the code hadn't been updated to reflect that.  It's also not super clear from the examples on how to use the protobuffs, expanded documentation would be great!

erinxocon, 2017-05-31:
I've written PR #75 to help address some of these issues!

samwel89, 2017-07-12:
Hi @erinxocon,

I have tested your PR but I still get some errors. Do I miss something? Or there must be some updates from Matrix Creator to be able to use your PR? However, I have also tested the original file "test_everloop.py" but I have this same error. Any idea?

Thank you in advance.

[image]

erinxocon, 2017-07-13:
huh interesting @samwel89, what version of python were you using?  I assume you have compiled the most up to date protobufs?

samwel89, 2017-07-13:
I use python 2.7.9, and my protoc --version is `libprotoc 3.2.0`

erinxocon, 2017-07-13:
@samwel89 I can take a look at it tomorrow!

samwel89, 2017-07-13:
@erinxocon Okay, I will be waiting! Thank you in advance.

samwel89, 2017-07-17:
Hello @erinxocon,
Any updates? Did you take a look?

erinxocon, 2017-07-17:
@samwel89 did you happen to check the new release containing the official python updates along with the updated protodrivers?  It can be found [here](https://github.com/matrix-io/matrix-creator-malos/tree/master/src/python_test).  You should just be be able to fetch the new changes!  Can you check those out and let me know?

samwel89, 2017-07-17:
Yes, I have just seen them after I texted you. However, I have tested "test_everloop.py" and I still get this error `ImportError: No module named matrix_io.proto.malos.V1` it looks like they didn't update the README.md to show how to import these new protodrivers. Any idea?

N.B:Before testing I have done: sudo apt-get update && sudo apt-get upgrade

erinxocon, 2017-07-17:
I've got some time right now, let me see if I can replicate.  I'm gonna start with a fresh image and go from there.

samwel89, 2017-07-17:
OK, cool! let me know your progress!

hpsaturn, 2017-07-17:
@samwel89 please try upgrade all pip packages on your raspberryPi:
`sudo pip freeze --local | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 sudo pip install -U`
and then install from requirements:
`sudo pip install -r requirements.txt`

samwel89, 2017-07-18:
@hpsaturn yes, now it works! Thank you.

closed 2017-07-18 by hpsaturn