matrix-io/matrix-creator-malos #78: apt get returns error

MexXxo, 2017-06-02:
Doing and apt-get update works fine, but the upgrade seems to be problematic:  (IP v6 changed because I'm not sure if it's mine)

Error http://packages.matrix.one/matrix-creator/ ./ matrix-creator-init 0.2-23
  403  Forbidden [IP: 2a00:1450:4005:80x::xxxx 80]
Error http://packages.matrix.one/matrix-creator/ ./ matrix-creator-malos 0.1-4.9
  403  Forbidden [IP: 2a00:1450:4005:80x::xxxx 80]

MexXxo, 2017-06-05:
Opening it with my browser give me this information:
`<Error>
<Code>AccessDenied</Code>
<Message>Access denied.</Message>
<Details>
Anonymous users does not have storage.objects.get access to object packages.matrix.one/matrix-creator/matrix-creator-init_0.2-23_armhf.deb.
</Details>
</Error>`

Bug also reported here, but there seems no acitivity:
[http://community.matrix.one/t/matrix-apt-get-packages-are-not-reachable-404-not-found](url)

kdpatino, 2017-06-05:
Hello @MexXxo,

You are using a old version of matrix-creator-init. Check the release info https://github.com/matrix-io/matrix-creator-init/releases/tag/0.3.0 

So you could fix with: `sudo apt-get update` and  `sudo apt-get upgrade` and then `sudo apt-get install matrix-creator-init`. 

Let me know if you have any question.

Regards,

Kevin

MexXxo, 2017-06-05:
Still the issue persist that I can not download or upgrade anything. Returns 403 (probably because anonymous user)

MexXxo, 2017-06-05:
Here are the logs from the terminal (it's in german, but I think it should be understandable)
sudo apt-get install matrix-creator-init
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Die folgenden NEUEN Pakete werden installiert:
  matrix-creator-init
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Es müssen 280 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 1.214 kB Plattenplatz zusätzlich benutzt.
WARNUNG: Die folgenden Pakete können nicht authentifiziert werden!
  matrix-creator-init
Diese Pakete ohne Überprüfung installieren? [j/N] J
Fehl http://packages.matrix.one/matrix-creator/ ./ matrix-creator-init 0.2-23
  403  Forbidden [IP: 2a00:1450:4005:80a::2010 80]
E: Fehlschlag beim Holen von http://packages.matrix.one/matrix-creator/./matrix-creator-init_0.2-23_armhf.deb  403  Forbidden [IP: 2a00:1450:4005:80a::2010 80]

E: Einige Archive konnten nicht heruntergeladen werden; vielleicht »apt-get update« ausführen oder mit »--fix-missing« probieren?

kdpatino, 2017-06-05:
Hello @MexXxo ,

Could you try with the Following steps:

```
sudo apt-get remove --purge matrix-creator-init
sudo apt-get update
sudo apt-get upgrade

sudo apt-get install matrix-creator-init
```

MexXxo, 2017-06-06:
Finally it works, maybe this would be helpful for further upgrades. 
The remove with the purge flag was the missing step :)

Thanks for your help!

closed 2017-06-06 by MexXxo