matrix-io/matrix-creator-malos #79: Mic DoA seems to be not updated

MexXxo, 2017-06-15:
Hi,

I observed that the doa seems to be not updated.
I get everytime the same values via ZMQ. 
If I'm restarting the malos service it is updated once... (I also changed the direction from where I was making noise on malos restart)

Can somebody please confirm, that this issue exist not only for me?

Here my terminal output:
node test_doa.js
Sending pings every 5 seconds
azimutal_angle (degrees) = 67.50000017081133
   polar_angle (degrees) = 36.00000100179127

azimutal_angle (degrees) = 67.50000017081133
   polar_angle (degrees) = 36.00000100179127

azimutal_angle (degrees) = 67.50000017081133
   polar_angle (degrees) = 36.00000100179127
^C
 sudo systemctl restart matrix-creator-malos.service
 node test_doa.js
Sending pings every 5 seconds
azimutal_angle (degrees) = 157.49999584510033
   polar_angle (degrees) = 72.00000200358254

azimutal_angle (degrees) = 157.49999584510033
   polar_angle (degrees) = 72.00000200358254

azimutal_angle (degrees) = 157.49999584510033
   polar_angle (degrees) = 72.00000200358254

azimutal_angle (degrees) = 157.49999584510033
   polar_angle (degrees) = 72.00000200358254

azimutal_angle (degrees) = 157.49999584510033
   polar_angle (degrees) = 72.00000200358254

MexXxo, 2017-06-19:
And what would be the 0 Degree LED ? So where would I need to calibrate to

stoneWeb, 2019-02-28:
I have same question! how to locate LED?