matrix-io/matrix-creator-malos #84: Mic Channels via ALSA not useable in pulseaudio

MexXxo, 2017-07-02:
Hi,

I just configured a pulseaudio server using the echo-cancel module.
As the microphones are added to ALSA as file streams, I could not figure out how to use them in pulseaudio.

Any suggestions? Would it be possible to tell ALSA and Pulseaudio the beamformed channel is a real microphone.

Nethertheless as Echo-Cancellation is on your roadmap, this would a good module to look at.

Hoeze, 2017-12-25:
I bought a Matrix Creator and I think about giving it back, as I cannot get it to work with PulseAudio.

huhlig, 2018-02-28:
Any update about when we will be able to use the mic's in pulseaudio or pyaudio?

kdpatino, 2018-02-28:
Hello @huhlig.

Yes, now we have a first implementation in the kernel space. It allows accessing the microphones with pyAudio
1-  https://github.com/matrix-io/matrixio-kernel-modules/blob/master/misc/pyaudio_test.py.
2- https://github.com/matrix-io/matrixio-kernel-modules
3- https://community.matrix.one/t/matrix-creator-and-google-assistant/1269/2?u=kevindpatino 

We are working on the support to the up layers (in this case MALOS). Now MALOS and kernel implementation is not compatible.

Regards,

Kevin Patino

0PK, 2018-02-28:
Great news, but are we reasonably looking at a solution in days, weeks or months? Any indication would be very much appreciated.

MexXxo, 2018-05-16:
years

kdpatino, 2018-05-16:
Hello @MexXxo,

We support MALOS with KErnel modules since April 10, check the release note:
https://github.com/matrix-io/matrix-creator-malos/releases/tag/v0.3.3

Thank You!! Let me know if you have any doubt or trouble with this update:

Best Regards,

Kevin Patino