matrix-io/matrix-creator-malos #87: Provide Roadmap

MexXxo, 2017-07-08:
In the last months, I could see that there is very few progress on the malos.
I think the malos part is the most important software part for all matrix devices.

I would like to see a kind of roadmap, what is still pending (like Z-Wave and ZigBee API) and some kind of estimation when you plan to work on it.

I totally understand that you're a small team and probably have a lot of work, so that it is only an estimation and probably it will take a while to get every part done.

This is my current view, what needs to be done:

- [ ] **Everloop** - Improved/Easier API 
- [ ] **Humidity** - Documentation and calibration of the values (e.g. Temperature)
- [ ] **IMU** 
- [ ] **IR** - No Receive API yet
- [ ] **Microphone Array** - Audio Driver + Improved DoA + [optional] Echo Cancellation
- [ ] **Pressure**  - Documentation and calibration of the values (e.g. Temperature)
- [ ] **UV** 
- [ ] **Zigbee** - No API yet
- [ ] **NFC** - No API yet
- [ ] **Z-wave** - No API yet
- [ ] **GPIO** - I have not tested this, maybe someone can provide a state to this?

MexXxo, 2019-07-01:
Hey guys,
what is the current state? 
Please close this if you won't provide a roadmap. 
I guess this would be also interesting for the matrix lite...

samreenislam, 2019-07-02:
Hi @MexXxo,

Thank you so much for following up on this post! 

We have been working on an updated roadmap, and will be posting it next week.

I'll link it here when it's public.

Best,
Samreen

MexXxo, 2019-07-14:
Hi @samreenislam,
short reminder for the roadmap :-)

samreenislam, 2019-08-06:
Hi @MexXxo,

[Here](https://community.matrix.one/t/matrix-development-checklist/2587) is an overall future feature development checklist that we plan to update over time.

In the next few days, we will also add mini checklists per major repo. Given the utility of MATRIX Lite, we will probably slow down development for MATRIX Core (MALOS). One of our ideas is to integrate our zigbee and ZWave modules into third party APIs like OpenHAB and OpenZWave.

Hope this helps!

Best,
Samreen