matrix-io/matrix-creator-malos #93: MalOS Java script test not functional anymore

mudij, 2017-07-27:
When I try to run any of the javascript tests under under js_test directory they fail looking for protocol-buffers files.

/home/pi/workspace/matrix-creator-malos/src/js_test/test_everloop.js:17
var matrixMalosBuilder = protoBuilder.build("matrix_malos")
                                     ^

TypeError: Cannot read property 'build' of null
    at Object.<anonymous> (/home/pi/workspace/matrix-creator-malos/src/js_test/test_everloop.js:17:38)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3

It seems that the following change has removed the git protocol-buffers sub modules for matrix-creator-malos.
https://github.com/matrix-io/matrix-creator-malos/commit/ef2ace87888be68844323f28357eaa1e5eb5ff2f

Are further changes required to fix the javasctipt tests?
If that is not the case can someone update the readme to show how to get the tests working again.
https://github.com/matrix-io/matrix-creator-malos/blob/master/README.md

mudij, 2017-07-27:
Just spotted some changes that are already on their way to address this issue.
https://github.com/matrix-io/matrix-creator-malos/pull/92/files

hpsaturn, 2017-07-28:
Right, the documentation is outdated and on hold for a new merge. For now please try the next flow:

```
echo "deb http://unstable-packages.matrix.one/ stable main" | sudo tee -a /etc/apt/sources.list
sudo apt-get update
sudo apt-get remove matrix-creator-malos --purge
sudo apt-get autoremove
sudo apt-get install matrix-creator-malos --no-install-recommends
sudo reboot
```

and update javascript samples directory and nodejs version if this outdated:

```
curl -sL https://deb.nodesource.com/setup_6.x | sudo bash -
sudo apt-get install nodejs
cd src/js_test/
rm -fr node_modules
```

maciekrb, 2017-08-05:
Hey @mudij we merged in the changes for the javascript examples. Please check them out, let us know if you find any issues.

mudij, 2017-08-08:
Great thanks, everything seems to be working again.

closed 2017-08-08 by mudij