matrix-io/matrix-creator-malos #104: Unable to start MALOS

henryoliver, 2018-04-10:
Hey guys, I'm getting an error on starting MALOS, appreciate any help here:

```
pi@raspberrypi:~ $ sudo systemctl status matrixio-malos
● matrixio-malos.service - Matrix Abstraction Layer for OS
   Loaded: loaded (/lib/systemd/system/matrixio-malos.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-04-10 14:48:29 EDT; 17min ago
  Process: 484 ExecStart=/usr/bin/malos (code=exited, status=127)
 Main PID: 484 (code=exited, status=127)

Apr 10 14:48:29 raspberrypi systemd[1]: Started Matrix Abstraction Layer for OS.
Apr 10 14:48:29 raspberrypi malos[484]: **************
Apr 10 14:48:29 raspberrypi malos[484]: MALOS starting
Apr 10 14:48:29 raspberrypi malos[484]: **************
Apr 10 14:48:29 raspberrypi malos[484]: /usr/bin/malos: symbol lookup error: /usr/bin/malos: undefined symbol: _ZN10matrix_hal11WishboneBusC1Ev
Apr 10 14:48:29 raspberrypi systemd[1]: matrixio-malos.service: Main process exited, code=exited, status=127/n/a
Apr 10 14:48:29 raspberrypi systemd[1]: matrixio-malos.service: Unit entered failed state.
Apr 10 14:48:29 raspberrypi systemd[1]: matrixio-malos.service: Failed with result 'exit-code'.
```

Thanks!

kdpatino, 2018-04-10:
Hello @henryoliver 

We just deploy a new release https://github.com/matrix-io/matrix-creator-malos/releases/tag/v0.3.3.
Could you run: `sudo apt update && sudo apt upgrade`

Regards,

Kevin Patino

henryoliver, 2018-04-10:
@kdpatino perfect! Worked!

```
pi@raspberrypi:~ $ sudo systemctl status matrixio-malos
● matrixio-malos.service - Matrix Abstraction Layer for OS
   Loaded: loaded (/lib/systemd/system/matrixio-malos.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-04-10 15:33:03 EDT; 7min ago
 Main PID: 557 (malos)
   CGroup: /system.slice/matrixio-malos.service
           └─557 /usr/bin/malos

Apr 10 15:33:03 raspberrypi malos[557]: can't open any device
Apr 10 15:33:03 raspberrypi malos[557]: You can query specific driver info using port 20012.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver IMU with port 20013.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver Humidity with port 20017.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver Everloop with port 20021.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver Pressure with port 20025.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver UV with port 20029.
Apr 10 15:33:03 raspberrypi malos[557]: INFO: Microphone Array Driver => Kernel Modules has been loaded. Use ALSA implementation
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver Servo with port 20045.
Apr 10 15:33:03 raspberrypi malos[557]: Registered driver Gpio with port 20049.
```

kdpatino, 2018-04-10:
Could

henryoliver, 2018-04-10:
@kdpatino I follow all the other steps through, but `node test_everloop.js` test is not working. I have the Matrix Voice device. How can I do a simple test by myself?

I don't know if it influences, but Im getting the first message below:
`Apr 10 15:52:31 raspberrypi malos[565]: can't open any device`

henryoliver, 2018-04-10:
@kdpatino yes, I installed first the `matrixio-kernel-modules` and then `matrixio-malos`

kdpatino, 2018-04-10:
Yes, That was the reason that I reopen the issue. 

Could you run and check:

```
dpkg -l | grep matrix
ii  libmatrixio-creator-hal:armhf   0.3.1                        armhf        Hardware Abstraction Layer for MATRIX.
ii  libmatrixio-malos:armhf         0.1.2                        armhf        Libraries of MATRIX abstraction layer for MOS
ii  libmatrixio-protos:armhf        0.1.4                        armhf        Protocolbuffers for MATRIX abstraction layer
ii  matrixio-creator-init           0.4.8                        armhf        Install scripts that can program the MATRIX Creator FPGA and SAM3 IMU.
ii  matrixio-kernel-modules         0.1-2                        armhf        Install kernel modules for MATRIX devices.
ii  matrixio-malos                  0.3.3                        armhf        Allows queries to MATRIX Creator hardware via ZMQ.
ii  matrixio-openocd                1.0.1                        armhf        Open on-chip JTAG debug solution for ARM and MIPS systems
ii  matrixio-protobuf               3.4.0                        armhf        Protocol Buffers - Google's data interchange format. (From GRPC 3.4.0)
ii  matrixio-xc3sprog               1.1.1                        armhf        Spartan3, XCF and CPLD JTAG programmer and other utilities.
```
Also, could you check:
```
uname -a
Linux matrixTest 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```

henryoliver, 2018-04-10:
```
pi@raspberrypi:/srv/app $ dpkg -l | grep matrix
ii  libmatrixio-creator-hal:armhf   0.3.1                        armhf        Hardware Abstraction Layer for MATRIX.
ii  libmatrixio-malos:armhf         0.1.2                        armhf        Libraries of MATRIX abstraction layer for MOS
ii  libmatrixio-protos:armhf        0.1.4                        armhf        Protocolbuffers for MATRIX abstraction layer
ii  matrixio-creator-init           0.4.8                        armhf        Install scripts that can program the MATRIX Creator FPGA and SAM3 IMU.
rc  matrixio-kernel-modules         0.1-2                        armhf        Install kernel modules for MATRIX devices.
ii  matrixio-malos                  0.3.3                        armhf        Allows queries to MATRIX Creator hardware via ZMQ.
ii  matrixio-openocd                1.0.1                        armhf        Open on-chip JTAG debug solution for ARM and MIPS systems
ii  matrixio-protobuf               3.4.0                        armhf        Protocol Buffers - Google's data interchange format. (From GRPC 3.4.0)
ii  matrixio-xc3sprog               1.1.1                        armhf        Spartan3, XCF and CPLD JTAG programmer and other utilities.
```

```
pi@raspberrypi:/srv/app $ uname -a
Linux raspberrypi 4.14.33-v7+ #1109 SMP Tue Apr 10 17:28:38 BST 2018 armv7l GNU/Linux
```

kdpatino, 2018-04-10:
Could you run:
```
lsmod | grep matrixio
matrixio_pcm          278528  1
matrixio_codec         16384  1
snd_soc_core          188416  2 matrixio_pcm,matrixio_codec
matrixio_uart          16384  0
matrixio_env           16384  0
matrixio_imu           16384  0
industrialio           73728  2 matrixio_imu,matrixio_env
matrixio_gpio          16384  0
matrixio_everloop      16384  0
matrixio_regmap        61440  1
matrixio_core          16384  4 matrixio_regmap,matrixio_pcm,matrixio_imu,matrixio_env
snd_pcm                98304  5 matrixio_pcm,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core,snd_dummy
```

henryoliver, 2018-04-10:
`lsmod | grep matrixio`
empty! Nothing in response to the command above.

kdpatino, 2018-04-10:
That's is the issue.
Run: 
1- Remove kernel modules from debian package:
 ```
$ sudo apt remove --purge matrixio-kernel-modules
$ sudo apt update
$ sudo apt upgrade
```
2- check if everything is removed
```
$ ls /lib/modules/4.14.30-v7+/kernel/drivers/mfd/ | grep matrixio
# If here exists any file run:
$ sudo rm  /lib/modules/4.14.30-v7+/kernel/drivers/mfd/matrixio*

ls /boot/overlays | grep matrixio
# If here exists any file run:
$ sudo rm /boot/overlays/matrixio*

$ sudo reboot
```
3- Try to install again the matrixio-kernel-modules
```
sudo apt matrixio-kernel-modules

sudo reboot 

$ more /etc/modules-load.d/matrixio.conf 
matrixio-regmap
matrixio-everloop
matrixio-gpio
matrixio-imu
matrixio-env

$ lsmod | grep matrixio
matrixio_pcm          278528  1
matrixio_codec         16384  1
snd_soc_core          188416  2 matrixio_pcm,matrixio_codec
matrixio_uart          16384  0
matrixio_env           16384  0
matrixio_imu           16384  0
industrialio           73728  2 matrixio_imu,matrixio_env
matrixio_gpio          16384  0
matrixio_everloop      16384  0
matrixio_regmap        61440  1
matrixio_core          16384  4 matrixio_regmap,matrixio_pcm,matrixio_imu,matrixio_env
snd_pcm                98304  5 matrixio_pcm,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core,snd_dummy
```

henryoliver, 2018-04-10:
Got an error on the new installation:

```
pi@raspberrypi:~ $ sudo apt install matrixio-kernel-modules
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dkms
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  dkms matrixio-kernel-modules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/88.4 kB of archives.
After this operation, 380 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package dkms.
(Reading database ... 74798 files and directories currently installed.)
Preparing to unpack .../archives/dkms_2.3-2_all.deb ...
Unpacking dkms (2.3-2) ...
Selecting previously unselected package matrixio-kernel-modules.
Preparing to unpack .../matrixio-kernel-modules_0.1-2_armhf.deb ...
Unpacking matrixio-kernel-modules (0.1-2) ...
Setting up dkms (2.3-2) ...
Setting up matrixio-kernel-modules (0.1-2) ...
Enabling firmware loading at startup

Creating symlink /var/lib/dkms/matrixio/0.1/source ->
                 /usr/src/matrixio-0.1

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.14.33-v7+ cannot be found at
/lib/modules/4.14.33-v7+/build or /lib/modules/4.14.33-v7+/source.
dpkg: error processing package matrixio-kernel-modules (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for man-db (2.7.6.1-2) ...
Errors were encountered while processing:
 matrixio-kernel-modules
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

kdpatino, 2018-04-10:
Ok. Try with: `sudo apt install raspberrypi-kernel-headers raspberrypi-kernel`

henryoliver, 2018-04-10:
```
pi@raspberrypi:~ $ sudo apt install raspberrypi-kernel-headers raspberrypi-kernel
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20180328-1).
raspberrypi-kernel-headers is already the newest version (1.20180328-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up matrixio-kernel-modules (0.1-2) ...
Enabling firmware loading at startup
Error! DKMS tree already contains: matrixio-0.1
You cannot add the same module/version combo more than once.
dpkg: error processing package matrixio-kernel-modules (--configure):
 subprocess installed post-installation script returned error exit status 3
Errors were encountered while processing:
 matrixio-kernel-modules
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

henryoliver, 2018-04-10:
```
pi@raspberrypi:~ $ sudo apt install raspberrypi-kernel-headers raspberrypi-kernel
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20180328-1).
raspberrypi-kernel-headers is already the newest version (1.20180328-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ sudo apt install matrixio-kernel-modules
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dkms
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  dkms matrixio-kernel-modules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/88.4 kB of archives.
After this operation, 380 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package dkms.
(Reading database ... 74800 files and directories currently installed.)
Preparing to unpack .../archives/dkms_2.3-2_all.deb ...
Unpacking dkms (2.3-2) ...
Selecting previously unselected package matrixio-kernel-modules.
Preparing to unpack .../matrixio-kernel-modules_0.1-2_armhf.deb ...
Unpacking matrixio-kernel-modules (0.1-2) ...
Setting up dkms (2.3-2) ...
Setting up matrixio-kernel-modules (0.1-2) ...
Enabling firmware loading at startup

Creating symlink /var/lib/dkms/matrixio/0.1/source ->
                 /usr/src/matrixio-0.1

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.14.33-v7+ cannot be found at
/lib/modules/4.14.33-v7+/build or /lib/modules/4.14.33-v7+/source.
dpkg: error processing package matrixio-kernel-modules (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for man-db (2.7.6.1-2) ...
Errors were encountered while processing:
 matrixio-kernel-modules
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

eljuguetero, 2018-04-10:
Please share the kernel version:

`uname -a
`
This drivers only works with current stock raspbian kernel. To reverting back to current stock Raspbian kernel (From https://www.raspberrypi.org/documentation/linux/kernel/updating.md)
`
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
`

mhermosi, 2020-12-09:
Sorry to re-open this but I have this:

it is raspbian updated/upgraded:
```
pi@myhost:~ $ lsb_release -sc
stretch
pi@myhost:~ $ uname -a
Linux myhost 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
```

the Service status:
```
pi@myhost:~ $ systemctl status matrixio-malos
● matrixio-malos.service - Matrix Abstraction Layer for OS
   Loaded: loaded (/lib/systemd/system/matrixio-malos.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2020-12-08 17:25:05 EST; 2h 35min ago
  Process: 831 ExecStart=/usr/bin/malos (code=exited, status=0/SUCCESS)
 Main PID: 831 (code=exited, status=0/SUCCESS)

Dec 08 17:25:04 myhost systemd[1]: Started Matrix Abstraction Layer for OS.
Dec 08 17:25:05 myhost malos[831]: **************
Dec 08 17:25:05 myhost malos[831]: MALOS starting
Dec 08 17:25:05 myhost malos[831]: **************
Dec 08 17:25:05 myhost malos[831]: can't open any device
```

the packages installed.
```
pi@myhost:~ $ dpkg -l | grep matrix
ii  libmatrixio-creator-hal:armhf         0.3.8                             armhf        Hardware Abstraction Layer for MATRIX.
ii  libmatrixio-malos:armhf               0.1.4                             armhf        Libraries of MATRIX abstraction layer for MOS
ii  libmatrixio-protos:armhf              0.1.6                             armhf        Protocolbuffers for MATRIX abstraction layer
ii  libswresample2:armhf                  7:3.2.15-0+deb9u1                 armhf        FFmpeg library for audio resampling, rematrixing etc. - runtime files
ii  matrixio-creator-init                 0.4.17                            armhf        Install scripts that can program the MATRIX Creator FPGA and SAM3 IMU.
ii  matrixio-malos                        0.3.6                             armhf        Allows queries to MATRIX Creator hardware via ZMQ.
ii  matrixio-openocd                      1.0.1.002                         armhf        Open on-chip JTAG debug solution for ARM and MIPS systems
ii  matrixio-protobuf                     3.8.0                             armhf        Protocol Buffers - Google's data interchange format. (From GRPC 3.8.0)
ii  matrixio-xc3sprog                     1.1.1.003                         armhf        Spartan3, XCF and CPLD JTAG programmer and other utilities.
```

When trying to install the kernel modules I am not able to do it neither the raspberry kernel headers:
```
pi@myhost:~ $ sudo apt install  matrixio-kernel-modules
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 matrixio-kernel-modules : Depends: raspberrypi-kernel-headers but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Please advise

-Martin