matrix-io/matrix-creator-malos #111: Problem with wakeword demo on Matrix Voice

nmstoker, 2018-07-08:
I am having trouble getting the wakeword example to work on my new Matrix Voice.

After receiving the device, I went through the initial installation for Matrix Core [https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/core-installation/](as described here) and was able to successfully get the Python LED demo working [https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/python-installation/](as here).

A few days later I then went to try to wakeword demo [https://matrix-io.github.io/matrix-documentation/matrix-core/python-examples/wakeword/](here) and it ran but did not respond at all to the wake words.  I then tried running the LED demo again but found that didn't work.

The only thing I had done in between the original LED demo and wakeword install was to do the apt-get update/upgrade as described in the "Helpful Information" "Upgrading" section of the install page.

I was stuck for a bit but then came across #104 and uninstalled things and reinstalled.  The steps worked as intended except the first time I ran `sudo apt remove --purge matrixio-kernel-modules` it gave an error:

```
  depmod....

  DKMS: install completed.
  Enable configurations in /boot/config.txt 
  Errors were encountered while processing:
  matrixio-kernel-modules
  E: Sub-process /usr/bin/dpkg returned an error code (1)`
```
Simply re-running `sudo apt remove --purge matrixio-kernel-modules` worked a second time.

The only slight change I needed to check the various things were removed was to update this line:

ls /lib/modules/**4.14.52-v7+**/kernel/drivers/mfd/ | grep matrixio

*(ie the current stock kernel - I haven't fiddled with that in any way, it's just what's installed with a regular setup)*

By that point the LED demo worked again.  However a second go at installing the wakeword does nothing and it seems like there may be some step missing.

Please could you advise if there's anything I should look at to diagnose this?

`uname -a` outputs:

**Linux rpi3Bplus1 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux**

Thanks,
Neil

eviathan, 2018-07-20:
I am having an identical problem to this. Literally everything else is working. Do we need to do some sort of audio configuration?

The c++ recording examples work fine. All the led and sensor examples work fine. 

The python wake word one just prints "listening for wakewords and does nothing" (I have configured the dic and lm files).

An odd thing to mention is that if you press ctrl+c and keyboard interrupt it goes onto printing 

"^CException KeyboardInterrupt in <bound method Context.__del__ of <zmq.sugar.context.Context object at 0x75e85c30>> ignored
Connected to error publisher with port 60005
Connected to data publisher with port 60007"

Which feels like its not connecting till you interupt it. I have also tried spinning up something to pull any data back from 0MQ and nothing.

Hermitter, 2018-07-20:
Thanks for the heads up! I've updated the [wakeword.py](https://github.com/matrix-io/matrix-core-examples/blob/master/python/wakeword.py) example. The issue seems to have been with how the example was sending its message to the [MATRIX CORE Base port](https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/understanding-core/#ports-protocol). Let us know if the updated example resolves your issue.

eviathan, 2018-07-23:
I tried running it and it did get further but it still isn't responding to the wakeword.

Hermitter, 2018-07-23:
@eviathan can you try running `ps aux | grep malos` on your Raspberry Pi? The output should be something similar to what I've pasted below. 
```
root       518  0.1  0.8 665764  7872 ?        Ssl  17:15   0:00 /usr/bin/malos
root       519  0.0  0.6 130744  5768 ?        Ssl  17:15   0:00 /usr/bin/malos_wakeword
root      1042  0.0  0.0   1900   404 ?        S    17:16   0:00 sh -c malos_psphinx -keyphrase "" -kws_threshold 1e-20 -dict "/home/pi/python-matrix-core-app/1339.dic" -lm "/home/pi/python-matrix-core-app/1339.lm" -inmic yes 2> /dev/null
root      1044 24.0  1.3  15720 12916 ?        S    17:16   1:43 malos_psphinx -keyphrase  -kws_threshold 1e-20 -dict /home/pi/python-matrix-core-app/1339.dic -lm /home/pi/python-matrix-core-app/1339.lm -inmic yes
pi        1135  0.0  0.0   4372   580 pts/0    S+   17:23   0:00 grep --color=auto malos
```

If the `malos_wakeword` and `malos pshinx` processes aren't running, try reinstalling malos_wakeword with `sudo apt install matrixio-malos-wakeword`

Hermitter, 2018-07-23:
Glad to be of help. @nmstoker are you still having trouble getting the wake word demo to work?

nmstoker, 2018-07-23:
I'll give the latest update a go. Been away, so haven't had time to check
it yet but thanks for looking into it.

nmstoker, 2018-08-11:
Hi @Hermitter - sorry for delay.  I'm having a look now, but I see no reference to  **malos_psphinx** in my process list.  How is that meant to be started (none of the steps I went through via the main documentation indicated I need to start it, but perhaps it's triggered by the wakeword.py script somehow?)

nmstoker, 2018-08-11:
Some more info since the above...  If I try running malos_pshinx using the equivalent of the command from the process list I get an ALSA error.

I try this:
````
malos_psphinx -keyphrase "" -kws_threshold 1e-20 -dict "/home/neil/matrix-voice/9017.dic" -lm "/home/neil/matrix-voice/9017.lm" -inmic yes
````
And after a load of config/start up info, I see

> INFO: malos_psphinx.cpp(168): malos_psphinx COMPILED ON: Jun  5 2018, AT: 17:00:53
> 
> ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
> Error opening audio device default for capture: No such file or directory
> FATAL: "malos_psphinx.cpp", line 105: Failed to open audio device 

I had't done anything with ALSA settings before, it was a fresh install of Raspbian and there wasn't anything mentioned in the installation (_right??_)

Hermitter, 2018-08-13:
Hey @nmstoker 

After installation, the PocketSphinx process should be running automatically on boot. You don't need to edit the ALSA settings.

One possibility for the issues you're encountering is if an error occurred during the installation process. I recommend starting from a fresh [Raspbian Stretch](https://www.raspberrypi.org/downloads/raspbian/) image and installing [MATRIX CORE](https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/core-installation/) & then [MATRIX Wakeword](https://matrix-io.github.io/matrix-documentation/matrix-core/protocols/wakeword/#installation).

If you're interested, we also provide a [setup guide](https://www.hackster.io/matrix-labs/matrix-voice-and-matrix-creator-running-snips-ai-b48344) for installing the [Snips.ai](https://snips.ai/) offline voice assistant.

 Let us know if you have any more questions or are still experiencing issues with the MATRIX Wakeword installation.

nmstoker, 2018-08-13:
I'll have a go as you suggest, but I'll need to finish it off tomorrow evening (it's late here now!).

Thanks a lot for your help!  I wonder if there's something about your installation steps that depends on the standard "pi" user being present? Changing that to another username is the only significant thing I'd done to my install (beyond the necessary stuff like setting the wifi details, hostname, enabling ssh).  And are there any issues using Jessie Lite?

Hermitter, 2018-08-13:
Are you using Raspbian Jessie for this installation? The MATRIX Kernel modules are built for Raspbian Stretch. Installing them on Raspbian Jessie will not work.

If that's the case, starting from a fresh Raspbian Stretch installation should fix the issues you're encountering.

nmstoker, 2018-08-13:
Sorry, my bad, I meant Stretch Lite! Was Stretch Lite before too :smile:

Hermitter, 2018-08-13:
Almost thought we found the issue you were having!

Yes, both Raspbian Stretch Lite & Desktop are compatible. I just tested this today from a fresh Stretch Lite installation. I didn't change the "pi" user, but that shouldn't have affected this installation process.

As a precaution, check the final console logs for each command you use to ensure the dependencies installed correctly.

nmstoker, 2018-08-14:
It worked! (I pushed on with it tonight rather than get half way and miss a step!)

I did have to run through the Python getting started steps too (to get the necessary modules) https://matrix-io.github.io/matrix-documentation/matrix-core/getting-started/python-installation/ but that was obvious.

Am unsure what the difference could have been - I suppose that something done during the period when the code wasn't working may have had some lasting effect. I agree that the user "pi" seems an unlikely cause.

One funny aspect is I have to put on a US accent to avoid my British "matrix start" being interpretted as "matrix stop" but I'll experiment with the sphinx settings (or get a voice coach! :wink: )

Thanks again!

Hermitter, 2018-08-14:
No problem. Glad to hear you got everything working!